ska-sa/astrokat #1: Add timing information to observation config file

rubyvanrooyen, 2018-10-02:
Timing items to add to configuration. This will be global and for the observation
 -- min_duration (required)
 -- max_duration
 -- (?)expected duration(?)

Should minimum observation time be specified by the requesting or staff astronomer?
Expected duration is specified in the SB, but dependent on dry-run to get the value, so perhaps this could be explicit?

Maximum duration is optional, if not specified,
the observation will loop while targets/LST is valid (in which case we probably need a --single tag)
Or
only perform a single pass through all the targets

spassmoor, 2018-10-04:
The min_duration  should be for by the requesting  scientist in the observation config file.

expected_duration is how long we expect to to run for based on simulation or dry run . 

max_duration should be by the AOD or person making the schedule. 
max_duration should equal expected_duration if it has been scheduled correctly .

rubyvanrooyen, 2018-10-23:
min_duration and obs_duration implemented, where the min_duration is the minimum limit for good science set by the astronomer and obs_duration is the expected observation duration simulated in the dry-run
I am not going to implemented a max_duration now since there is now clear implementation and therefore no clear requirement or test case

closed 2018-10-23 by rubyvanrooyen