ska-sa/astrokat #2: command line execution of astrokat-observe.py

rubyvanrooyen, 2018-10-02:
Offline the "simulated" environment is activated by the user command:
python astrokat-observe.py --profile test_targets.yaml

Obviously, this will cause issues when on the live system which will try to trigger a CAM configuration

Do we need to worry about disabling this on the live system, or simply assume it will be okay since the user should technically never see obs.mkat (I am not feeling happy about this assumption)

rubyvanrooyen, 2018-10-11:
On local system:
python astrokat-observe.py -o ruby --template ../input/SCI-20180516-PW-01.yaml
Using the observer.datetime option:
`2018-10-11 09:28:39Z - Total observation time 5160.00 seconds`
Using the datetime.now() option:
`2018-10-11 09:28:37Z - Total observation time 0.02 seconds`

On devcomm system:
python astrokat-observe.py -o ruby --template /home/kat/usersnfs/framework/extras/SCI-20180516-PW-01.yaml --sb-id-code=20181011-0003 --dry-run True

Timing issue raised separately in issue #9 

First only ran on command line -- got the following error

```
kat@monctl.devcomm.camlab.kat.ac.za:~/usersnfs/framework/astrokat/scripts$ python astrokat-observe.py -o ruby --template /home/kat/usersnfs/framework/extras/SCI-20180516-PW-01.yaml --sb-id-code=20181011-0003 --dry-run True
WARNING:katcorelib:Connecting to device katpool. Attempting to initialise...
WARNING:katcorelib:Connecting to device sched. Attempting to initialise...
log conf: katcamconfig/log/log.devcomm.conf
logger.name: katcorelib
2018-10-11 14:50:52.783Z kat.katcorelib.utility INFO sb_is_active(utility.py:719) sb_is_active INFO: Before get_value() active_schedule_status=nominal value= @ 1539269452.64,1539269452.03
2018-10-11 14:50:52.787Z kat.katcorelib.utility INFO sb_is_active(utility.py:736) sb_is_active INFO: active_schedule_status=nominal value= @ 1539269452.79,1539269452.03
2018-10-11 14:50:52.787Z kat.katcorelib.utility WARNING sb_is_active(utility.py:743) sb_is_active FALSE: sb_id_code=20181011-0003
2018-10-11 14:50:52.792Z ERROR    Could not build KAT container for site=devcomm_mkat system=katcamconfig/systems/devcomm_mkat.conf (Failed to build KAT container for schedule block 20181011-0003 (Could not retrieve dry-run allocations of for IN-ACTIVE SB 20181011-0003))
Unhandled exception <type 'exceptions.ValueError'> : Could not build KAT container for site=devcomm_mkat system=katcamconfig/systems/devcomm_mkat.conf (Failed to build KAT container for schedule block 20181011-0003 (Could not retrieve dry-run allocations of for IN-ACTIVE SB 20181011-0003))
Traceback (most recent call last):
  File "astrokat-observe.py", line 483, in <module>
    with telescope(opts, args_) as mkat:
  File "astrokat-observe.py", line 188, in __init__
    self.array = verify_and_connect(opts)
  File "/usr/local/lib/python2.7/dist-packages/katcorelib/observe.py", line 295, in verify_and_connect
    raise ValueError(msg)
Running cleanup for sched
Running cleanup for katpool
```

Did a dry-run using the GUI and then repeated the command, which then succeeded ?

On live system:

http://10.97.1.13:8081/tailtask/20181011-0012/dryrun
http://10.97.1.13:8081/tailtask/20181011-0012/progress

obs.sb.new(owner='Thunderkat')
obs.sb.type=katuilib.ScheduleBlockTypes.OBSERVATION
obs.sb.antenna_spec='available'
obs.sb.controlled_resources_spec='cbf,sdp'
obs.sb.description="Thunderkat: XRB"
obs.sb.instruction_set="run-obs-script /home/kat/usersnfs/ruby/astrokat/scripts/astrokat-observe.py -o lindsay --template /home/kat/usersnfs/framework/extras/SCI-20180516-PW-01.yaml --horizon=20 --proposal-id='SCI-20180516-PW-01' "
obs.sb.to_defined()
obs.sb.to_approved()
obs.sb.unload()

rubyvanrooyen, 2018-10-23:
@ajoubertza and @wbode-za : I updated the implementation on devcomm with the newer cleaner codebase and some updated from the users
When I tried to run the above command again the command line dry-run showed different error that may indicate another thing I have not thought about before...

`kat@monctl.devcomm.camlab.kat.ac.za:~/usersnfs/framework/astrokat/scripts$ python astrokat-observe.py -o ruby --template /home/kat/usersnfs/framework/astrokat/test/test_targets.yaml --horizon=20 --proposal-id='ENG-new-devel-test' --dry-run True`

When trying to find the sensor for `sub_band` the following error was obtained and this persists even after a GUI dry-run
```
Unhandled exception <type 'exceptions.AttributeError'> : 'NoneType' object has no attribute 'get_value'
Traceback (most recent call last):
  File "astrokat-observe.py", line 487, in <module>
    with telescope(opts, args_) as mkat:
  File "astrokat-observe.py", line 200, in __enter__
    self.subarray_setup(self.array, self.opts.template['instrument'])
  File "astrokat-observe.py", line 257, in subarray_setup
    sub_sensor = mkat.sensor.get(sensor_name).get_value()
Running cleanup for sched
Running cleanup for katpool
```

Can it be that the script simply does not know which subarray it is associated with? Which is why the command line dry-run seems to fail from time to time?

ajoubertza, 2018-10-23:
Yes, the script won't know which subarray it is associated with.  CAM dry runs can only be done when an SB in assigned to a subarray, and the execution will include the `--sb-id-code=` option - from this SB ID, CAM can determine the subarray.

rubyvanrooyen, 2019-04-26:
retiring this issue -- dry-run in OPT

closed 2019-04-26 by rubyvanrooyen