ska-sa/astrokat #5: Noise diode usage requirements

rubyvanrooyen, 2018-10-08:
I am trying to get some indication of noise diode implementation in typical observations. I would like to get some requirements for noise diode usage in order to work with CAM to updated the  [mkat-session.py](https://github.com/ska-sa/katcorelib/blob/master/katcorelib/mkat_session.py
) implementation

### Outdated usage instructions
New definition of noise diode usage from meerkat digitiser-cam-icd:
`?noise-source time on-fraction cycle-length`
Current noise diode input option defined in Line 97 of [observe.py](https://github.com/ska-sa/katcorelib/blob/master/katcorelib/observe.py) is to set a noise diode pattern, but this is out of date and has to be updated.
Will there be any great objections to setting the default to `off` with the update?

### Sessions functionality updates
`Line 1235: def fire_noise_diode` (function out of date)
`Line 1408: def track` (affected by noise diode functionality)

Current understanding of noise diode usage
* The ability to trigger the noise diode between tracks for a specified number of seconds -- as per the current implementation in the `track` function -- is still desirable

* Current implementation for the new observation framework is that when a noise diode pattern is requested it will be set at the start of the observation and deactivated at the end of the observation
I know of one beamformer requirement that confirms the this implementation of the noise diode pattern, with the additional need to be able to deactivate the noise diode for selected target observations, and then to reactivate after target observation.
Current thoughts are to add a `no_nd` tag per target

spassmoor, 2018-10-09:
I think that the noise diode firing in def track  should be removed as this complicates  the code and the deterministic nature of the observation, but then allowing a command to def fire_noise_diode in the framework so that it could be specified before  and/or  after a track length

mserylak, 2018-10-11:
Is this change in place now? Will older scripts work (i.e. is CAM backwards compatible)?

I have slightly different opinion that @spassmoor:
- I confirm that there are experiments (especially in beamformer) where you want ND firing during track,
- I can imagine that one would like to fire ND when observing with collecting imaging data (we certainly did that with BenH),
- I would also add continuous firing (@spassmoor, @rubyvanrooyen will that saturate correlator?)
- instead of `no_nd` tag we might want to use `nd` tag as there will be more observations without noise diode than with,

spassmoor, 2018-10-11:
Just a note about noise diode 'firing ... will that saturate correlator ' This is not an issue for the F & X engines as the noise diode is only adding 3 dB to the signal (doubling the signal power ) .  In the B engine, it is a bit different , but even then it is just adding 20 K to a system with a T_sys of ~6K  so maybe 6 dB increase.

rubyvanrooyen, 2018-11-02:
Noise diode implemented, will be tested and made to work
Functionality detail:
https://github.com/ska-sa/astrokat/wiki#noise-diode-usage-during-observations

closed 2018-11-02 by rubyvanrooyen