ska-sa/astrokat #18: Using katconfig calibrators in scripts

rubyvanrooyen, 2018-10-23:
Usage of the new implementation to read from katconfig

Using `astrokat-cals.py` I get the following error on devcomm when trying to read the file   
`katconfig/user/catalogues/Lband-gain-calibrators.csv`
```
Unhandled exception <type 'exceptions.AssertionError'> : Catalogue file does not exist in node config!
Traceback (most recent call last):
  File "astrokat-cals.py", line 541, in <module>
    main(cli(sys.argv[0]))
  File "astrokat-cals.py", line 468, in main
    observatory.read_file_from_node_config(cal_catalogue))
  File "/mnt/nfs/usersnfs/framework/astrokat/scripts/astrokat/observatory.py", line 88, in read_file_from_node_config
    assert katconf.resource_exists(catalogue_file), err_msg
```

rubyvanrooyen, 2018-10-23:
Possibility of also implementing this for the phase-up and delaycal scripts currently living in katsdpscripts, or should we keep these separate for the moment using only catalogues that is provided?

ajoubertza, 2018-10-23:
@lanceWilliams Have you checked if the new calibrator files are available on devcomm?

rubyvanrooyen, 2018-10-30:
offline systems, you need to specify the calibrator path:
`python astrokat-cals.py --prop-id 'SCI-propid-0' --pi 'No One' --contact 'dummy@ska.ac.za' --cal-tags gain bp pol flux delay --outfile ../output/catalogue_name.csv --infile ../input/sample_targetlist_for_cals.csv --datetime '2018-08-06 12:34' --cat-path ../../katconfig/user/catalogues/`

devcomm, the files comes for katconfig (very nice)
`python astrokat-cals.py --prop-id 'SCI-prop-id' --pi 'No One' --contact 'dummy@ska.ac.za' --cal-tags gain bp pol flux delay --outfile /home/kat/usersnfs/framework/extras/catalogue_name.csv --infile /home/kat/usersnfs/framework/extras/sample_targetlist_for_cals.csv --datetime '2018-08-06 12:34'`

See implementation details on   
https://github.com/ska-sa/astrokat/wiki#migrating-imaging-observations-to-the-new-framework-a-simple-howto

closed 2018-10-30 by rubyvanrooyen