ska-sa/astrokat #32: build of astrokat

tony2heads, 2018-11-23:


tony2heads, 2018-11-23:
# Attempt to build on my laptop
* did  a 'git clone'
* got an error message attempting to build:

python setup.py build
/usr/lib/python2.7/distutils/dist.py:267: UserWarning: Unknown distribution option: 'install_requires'
  warnings.warn(msg)
running build
running build_py
file astrokat.py (for module astrokat) not found
file astrokat.py (for module astrokat) not found
 
* checked to see what is there:

14:29@astrokat>ls astrokat/
correlator.py  __init__.pyc  __main__.pyc   observatory.py   simulate.py   utility.py
__init__.py    __main__.py   noisediode.py  observatory.pyc  simulate.pyc  utility.pyc

rubyvanrooyen, 2018-11-23:
During the demo it was explicitly stated that the installable package will be delivered when CAM takes over and that when ready and merged into master the repo should only be cloned and will work out of the box.

closed 2018-11-23 by rubyvanrooyen