ska-sa/astrokat #108: AttributeError: 'Antenna' object has no attribute 'observer'

nhurleywalker, 2021-08-25:
I'm attempting to run astrokat-targets.py to find a gain calibrator:

`tash@GLaDOS:~$ python3 /usr/local/bin/astrokat-targets.py --target src 00:01:00 -01:00:02.31
Traceback (most recent call last):
  File "/usr/local/bin/astrokat-targets.py", line 4, in <module>
    __import__('pkg_resources').run_script('astrokat==0.1.dev773+master.3ca200e', 'astrokat-targets.py')
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 658, in run_script
    self.require(requires)[0].run_script(script_name, ns)
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 1438, in run_script
    exec(code, namespace, namespace)
  File "/usr/local/lib/python3.6/dist-packages/astrokat-0.1.dev773+master.3ca200e-py3.6.egg/EGG-INFO/scripts/astrokat-targets.py", line 1077, in <module>
    outfile=args.outfile,
  File "/usr/local/lib/python3.6/dist-packages/astrokat-0.1.dev773+master.3ca200e-py3.6.egg/EGG-INFO/scripts/astrokat-targets.py", line 856, in main
    observatory = Observatory()
  File "/usr/local/lib/python3.6/dist-packages/astrokat-0.1.dev773+master.3ca200e-py3.6.egg/astrokat/observatory.py", line 62, in __init__
    self.observer = self.get_observer(horizon=horizon)
  File "/usr/local/lib/python3.6/dist-packages/astrokat-0.1.dev773+master.3ca200e-py3.6.egg/astrokat/observatory.py", line 149, in get_observer
    observer = self.kat.observer
AttributeError: 'Antenna' object has no attribute 'observer'`

The error seems to be a fault in the instantiation of the Observatory() class? I'm not sure how to fix this.

ajoubertza, 2021-08-25:
Hi @nhurleywalker 

That looks like a problem with the version of katpoint you have installed.  katpoint < 1.0 still has an `observer` attribute on the `katpoint.Antenna` objects.  Do you maybe have katpoint `1.0a1` installed?  You could try with katpoint `0.10`.  The older versions of katpoint still use ephem, instead of astropy.

We should probably update astrokat to be compatible with both.

nhurleywalker, 2021-08-25:
To install the code, I cloned it from github and ran

`python setup.py install`

So if it has an outdated version of katpoint then it got that from the setup command. I see in the docs you actually recommend using pip to install directly from github, so I wonder if I had done that whether it would have worked properly. Tomorrow I'll see if I can test this or at least review the logs to see where it got katpoint from when it installed. Thanks for the prompt reply!