ska-sa/astrokat #120: drift scans

mariogrs, 2024-04-22:
Is the drift scan observation type properly implemented? I tried it and astrokat-observe gave some weird output with negative elevation:

2024-04-22 15:02:48Z - Setting up telescope for observation
2024-04-22 15:02:48Z - Observation start up
2024-04-22 15:02:48Z - Running astrokat version - 0.0+unknown.202404221703
2024-04-22 15:02:48Z - Skipping instrument checks - approved_schedule does not exist
2024-04-22 15:03:12Z - Found 1 target(s): 0 from 0 catalogue(s), 0 from default catalogue and 1 as target string(s)
2024-04-22 18:00:00Z - Observation targets are ['3C273']
2024-04-22 18:00:00Z - Set noise diode period to multiple of correlator dump period: cycle length = 19.5 [sec]
2024-04-22 18:00:00Z - Request: Set noise diode pattern to activate at 1713808812.0 (includes 12.0 sec lead time)
2024-04-22 18:00:00Z - Antennas found in subarray, setting ND: m011,m022,m033,m044
2024-04-22 18:00:00Z - Repeat noise diode pattern every 19.5 sec, with 0.585 sec on and apply pattern to ['m011', 'm022', 'm033', 'm044']
2024-04-22 18:00:12Z - Report: Switch noise-diode pattern on at 1713808812.0
2024-04-22 18:00:12Z - Waiting for observation setup
2024-04-22 18:00:15Z - INIT
2024-04-22 18:00:15Z - Slewing to first target
2024-04-22 18:01:00Z - Slewed to 3C273 at azel (61.0, 36.3) deg
2024-04-22 18:01:00Z - Tracked 3C273 for 0 seconds
2024-04-22 18:01:00Z - Initialising Drift_scan target 3C273 for 400.0 sec
2024-04-22 18:01:00Z - Drift_scan observation for 400.0 sec
2024-04-22 18:02:34Z - Slewed to Az: -107:47:45.3 El: -31:05:16.1 at azel (-107.8, -31.1) deg
2024-04-22 18:09:14Z - Tracked Az: -107:47:45.3 El: -31:05:16.1 for 400 seconds
2024-04-22 18:09:14Z - Observation list completed - ending observation
2024-04-22 18:09:14Z - Observation loop statistics
2024-04-22 18:09:14Z - Single run through observation target list
2024-04-22 18:09:14Z - Total observation time 554.70 sec (9.25 min)
2024-04-22 18:09:14Z - Targets observed :
2024-04-22 18:09:14Z - 3C273 observed for 400.0 sec
2024-04-22 18:09:14Z - Observation clean up
2024-04-22 18:09:14Z - Returning telescope to startup state
2024-04-22 18:09:14Z - Resetting all noise diodes to "off"

spassmoor, 2024-04-29:
Hi @mariogrs  
Could you attach or post and example yaml file for us to reproduce this error.
Thanks
Sean

mariogrs, 2024-04-29:
durations:
  start_time: 2024-04-22 18:00:00     # This is the start time so that you can test these on using astrokat

instrument:
  integration_time: 2

# set noise diode pattern to fire every 20 seconds and only be on for 0.6s
noise_diode:
  antennas: all
  cycle_len: 19.5   # The maximum time is 20 seconds
  on_frac: 0.030    # the time  the noise diode is on =  cycle_len* on_frac
  lead_time: 12.   #  do not change

horizon: 30
observation_loop:
  - LST:  07:00-18:00 
    target_list:
      - name=3C273,  radec= 12:29:6.6996 02:03:8.5968, tags=target, duration=400.0, type=drift_scan

spassmoor, 2024-04-30:
I have confirmed the presence of the bug